lzrtag/LZRTag #67: Document Ruby more!

Xasin, 2019-10-06:
The current documentation is sparse, up to non-existent
This is a bit of a problem as it will make it hard for other people to properly write their own games.
Further yet, our own work with the system is being halted due to needing to look up event names and such.

A good documentation of the Ruby API, events and MQTT Topics, as well as a proper Bundler list of required gems to run the system would be appropriate.

travisghansen, 2019-10-18:
Just looking into this project with a few others for fun...any interest in creating a docker image? I've not got much experience with ruby but I'll see what I can do if you're interested.

Xasin, 2019-10-18:
@travisghansen Thanks for your interest!
A docker image as quickstart could be quite interesting, actually, since this project is growing in a bit of complexity ^^'

However, if you only want to test out the Ruby backend, I will soon publish a first version of the software as a rubygem, which makes installation as easy as `gem install lzrtag-backend`

If you're patient, I'll be writing a guide on setting up the infrastructure soon :)
Otherwise, you can poke me via [Telegram](t.me/Xasin) or [HackADay](bit.ly/XASLZR) and I could help you set up your system now ^^

travisghansen, 2019-10-18:
We'll be working on it over the course of several months so no rush. I did manage to get gems installed but it was failing to launch (not sure if I was even launching it correctly). I noticed the mqtt server was hard-coded as well so I hacked that and eventually got it to connect but still was getting some failure that appeared to be bad syntax.

Also we started looking into printing the guns. Do you have any stl files that are just the parts needed for printing? We're not 3d modeling folks but the fusion 360 stuff looked to include way more that what should be printed.

Thanks!

Xasin, 2019-10-18:
@travisghansen There is definitely a bit too much Ruby code floating around right now that doesn't really do much. I'll move those into a "deprecated" folder.

Also, the current system isn't meant to just be a deployable server and more of a base code for other people to code their own games on. As such, you're supposed to write your own Ruby script that at the very least instantiates the MQTT server connection and a few games.

You can see a more functional test script under Ruby/LasertagBase/lib/lzrtag/test.rb, and as said, I am working on much better documentation right now :>

And I could definitely upload a few 3D model files.
Those would be put onto HackADay, and might require a bit of tweaking to get working - the current system is designed to be printed with a multi-material printer, but I do it by manually swapping filaments mid-print with a small helper script and Prusa Slic3r's SEMM function.

travisghansen, 2019-10-18:
Oh wow. That might be a bit advanced printing for us :) we'll see. I'll just wait on the Ruby front then.

Thanks!

Xasin, 2019-10-18:
@travisghansen 
The multi-material nature of the print is optional. Technically, the whole thing can be done with a single material too - the only downside would be that you'd lack the light diffusion on the side plates of the weapon, but you could just stick a piece of paper over that instead, or hot glue it :)

I don't have access to the computer with the model on it until Monday, but I'll send over the files then.

travisghansen, 2019-10-18:
Sounds great, no rush at all from our side.

Xasin, 2019-10-20:
Alright, closed by the latest PR #71 
There are a few examples and a better README on the library code now :>

closed 2019-10-20 by Xasin