lzrtag/LZRTag #77: No BOM - List in documentation.

FreakyFerrox, 2019-12-17:
Would you be so kind to add a Built of Material List. 
From the KiCad file I am not able to extract a usable Part/Component-List.

thx

Xasin, 2019-12-18:
Hi @FreakyFerrox!

Thanks for the reminder. I was going to get back to the Lasertag this Christmas holiday and I will then make a proper BOM, probably with MPN or DigiKey part numbers. 
Some of the modules aren't on LCSC sadly, but I can check.

For now, I have a pretty much complete BOM on the Aisler project, but since that can't be exported (I think) I can see where you're coming from.

Xasin, 2020-01-01:
@FreakyFerrox is the BOM still needed or were you able to figure something out?
Now that all the Christmas and new year stuff is over I'd have a bit of time to sort a list out. I will have to change a part though since the IR LED I was using is no longer available!

FreakyFerrox, 2020-01-09:
Hi Xasin!

Sorry for the delay, somehow I skippt the notification mail...

The BOM List is still needed. One of my friends tried to get a Part/Component List of out of the shematic with kiCAD but that didn't work well.

And we don't want to use Aisler, it's a little expensive... ^^

Xasin, 2020-01-11:
@FreakyFerrox I'll get to that this weekend, I hope. 
My Laptop broke down so first that'll be fixed <.<

Where will you be getting your components from! Everything I chose can be bought off of DigiKey, while LCSC does not necessarily have some parts (I checked, would've bought from there <.<)

Just as a word of warning ^^'

Xasin, 2020-01-12:
@FreakyFerrox I have a DigiKey parts list here.
I think I can make a Manufacturer Part Number as well, which might be better in the long run...

https://hackaday.io/project/160804-lzrtag-flexible-diy-lasertag/files