lzrtag/LZRTag #78: Unable to build ESP code

spectre3ooo, 2020-01-05:
I haven't been able to build the code for the ESP32.  I'm getting this error:
`In file included from ../main/main.cpp:22:0:`
`../main/core/setup.h:11:34: fatal error: xasin/BatteryManager.h: No such file or directory`

I am building on Windows, and I've also cloned the XasCode repo to 
`C:\Development\XasWorks\XasCode\`.  I set the `XASCODE_PATH` environment variable to this path.  I'm running `idf.py build` from within `ESP32/Main` when it returns that error.

I pasted the full console output of the build here: https://gist.github.com/spectre3ooo/de4ae4e4b37ab2f1d9d206532187189c

Any ideas?

Xasin, 2020-01-05:
I have a good idea - you are using the latest version of the ESP IDF, right?
I haven't been able to make my own libraries compile under the new compiler system, the last time I tried to get it to work it didn't recognize my files. 
Use the ESP IDF v3.3 and try again

I'll be porting my libraries soon though, now that the v4 IDF has been out for a while and *should* work.

spectre3ooo, 2020-01-05:
I'm using v3.3.1.  Should that work, or should I install v3.3?

Xasin, 2020-01-05:
@spectre3ooo v3.3.1 is also fine.
However, keep in mind that the build system changed - you now need to call "make" and "make flash" or "make monitor" to start the build process.

Also, try checking out to "dev_3.2" or "dev_3.2_stats"
I left the whole thing in a yet unmerged state since I got slightly overwhelmed by my University exams... Proooobably not the best idea.

spectre3ooo, 2020-01-06:
Good to know.  I've been trying to run `idf.py build` on the `dev_3.2_stats` branch.  I had followed the latest documentation to do the install, which wasn't working.  Once I went back and followed the v3.3.1 docs for the install it built successfully.  Thanks!

I've been attempting to mod these guns (http://amzn.com/B07JH3BP8C) with your system.  The hardware available is a little different, though - only one RGB LED used for team color, two simple red muzzle flash LEDs, reload button on bottom, etc.  I had to design a new PCB layout to fit where the old one went.  It's working pretty well so far, except for a couple errors I made with the schematic that will require a new board revision.

Xasin, 2020-01-06:
Glad to hear it's compiling now!

The fact that you made your own board for these guns is really awesome, by the way! If you have some pictures or similar I would love to see them.

If you want I could help you modify the code to have a changed number of LEDs along the body, or a different button input, but you seem to be managing well enough.

closed 2020-01-06 by Xasin

spectre3ooo, 2020-01-06:
Yeah, absolutely.  I'll upload some pics this evening.   
Help on the LEDs and buttons would be greatly appreciated.  I'll update my fork with my changes and pics, and add issues to describe the LED/button functionality changes.

Side note, what speaker did you use?  The one that came in the guns is sounding somewhat overdriven.

spectre3ooo, 2020-01-08:
I haven't had a chance to clean up the code/schematics to update my fork, but I did throw some images up on imgur here: https://imgur.com/a/twxOzUD

Xasin, 2020-01-08:
@spectre3ooo Take your time setting things up, I'll be rather busy with some Uni stuff for the next two weeks sooo ...

It's best if you open an issue about this on your fork and @ me whenever you're ready.

spectre3ooo, 2020-01-08:
No problem.  School is definitely more important.

The existing speaker looks like a common 29mm diameter.  Would these work?  http://amzn.com/B07FMR5JGX

These look almost identical, though: http://amzn.com/B07M8ZZTJS

Xasin, 2020-01-08:
Anything that is 8 Ohms and less than 2W will be fine for the MAX driver IC, and needs no further adjustment either :)