lzrtag/LZRTag #79: Question re: The PCB needs a redesign due to design flaws!

philippsandi, 2020-08-17:
Hey there, 
very nice project :-) 
I stumbled upon this caution above regarding the PCB and wanted to ask, if you could specify the existing design flaws?

Thanks and greetings,
Philipp

Xasin, 2020-08-17:
Ah, that might be because I still need to update the Aisler website!
I actually got through most of the redesign and tested a Revision 2 of the PCB, but due to my University getting busier I was so far unable to properly update things there.

The PCB is pretty much fully done, but if you are patient, I would like to give it a quick review in the next day or two to make sure I didn't forget anything :)

philippsandi, 2020-08-18:
Oh cool, thank you for the quick answer - I'm in no hurry, so I'll gladly wait for the Revision 2. :-)

Xasin, 2020-08-19:
No problem @philippsandi, it's always a pleasure to see people enjoying my work! 

I went over the PCB files, and I think I was able to round everything off already, so I think it's safe to use the current design files. The main problems were with a button that... I no longer have anyways <.<'

I recommend ordering the boards off of Aisler or JLCPCB, depending on how many you need.
Usually I'd say Aisler is OK, but right now their BOM cost is REALLY messed up - it's telling me 2000€ for one set of electronics for my Lasertag system, which used to be 30€...

I have a Digikey parts list I can give you though:
https://hackaday.io/project/160804-lzrtag-flexible-diy-lasertag/files

Will this be sufficient?