lzrtag/LZRTag #80: A bit lazy...

zwep, 2021-05-24:
Hi,

I am a little bit lazy/busy with other stuff, but would love to try this project out. Is there somewhere a pre-written `How-to` on how to use this repository? Which stuff to buy, and how to assemble everything?

If that is not the case, totally fine as well. But then I know that I could add something like that in the near future

Xasin, 2021-05-25:
Hi Zwep!
There isn't really a tutorial or similar for this right now.
The closest you can get is our write-up on HackADay found here: https://hackaday.io/project/160804-lzrtag-flexible-diy-lasertag

*However*, I would recommend waiting a little bit.
This project is due for a thorough clean-up, both in the code and hardware. If you can wait about half a year to a year, there will be a complete remake of most components to provide a much nicer base for people to work off of. 

The current version works fine, but it has a few flaws and quirks that I'm not quite happy with.

I'll let you decide if you want to wait or build one now, but I recommend waiting a bit :)

Ps: If you'd like to do a "user-test" later on and help me make good tutorials for how to program and assemble these, I would absolutely love that.
I could add you as a collaborator to the HackADay or GitHub page so you're in the loop.

zwep, 2021-05-26:
Hi Xasin,

ah cool! Nice to hear :) I think I'll wait for now and see if I can help later on with any user-testing. 

However, I am curious if you are going to change anything to the 3D printed case. I have a 3d printer at home which allows me to already assemble one casing for example :)

Greetings,
zwep

Xasin, 2021-05-26:
I actually wanted to do a complete redesign of the casing as well ^^

The new casing needs to include a proper mount for the speaker - the current version provides horrible sound, which is in part due to how I placed the speakers, and it's just such a shame.

It'd also be a lot bigger and offer more ways to mount more visually appealing LEDs, as well as a OLED display as heads-up-display and better buttons :>

You can still 3D print the current casing since it's not a lot of material, but I'd wait a bit.

The 3D printed case files should also be on HackADay

lucullusTheOnly, 2021-06-18:
Hi,

I've just found you at digital Maker Fair Hannover. Just as zwep I'm also in for user testing. In the past I tried to build such a system myself, though the overall structure lead into a deadlock, so I abandoned the project. Currently I don't have much time for contributing (my own projects filling much of the free time), but I will look further into your files. Maybe I can also contribute a bit.

Thanks for your work! I definitely will watch your progress :D

laserir, 2021-12-24:
Hey there! Hope everything is going well for you!

Mind to give us a quick update about the v4.0 hardware and software changes?

Xasin, 2021-12-30:
@laserir 
Thanks for your interest!
As of right now, the project is still on hold for a few reasons:
- I'm focusing on my academic studies and am progressing through my master's thesis
- The component shortages have hit a lot of the vital elements I would normally use, including battery monitors etc.
- I'm waiting for the ESP32-S3 to be more widely available on DigiKey or a similar site.

I also have a few personal projects that take precedence, but I am hoping to start working on a newer version of the Hardware somewhen during the first half of 2022. 
The list of changes I want to make has grown a little more, you can find it [here on HackADay](https://hackaday.io/project/160804-lzrtag-flexible-diy-lasertag/log/189364-current-progress-and-moving-forwards)

tmb088, 2022-03-24:
Is this software based on [https://openlasertag.org/language/en/what-is-openlasertag/](url)?

Xasin, 2022-03-24:
@tmb088 no, this entire ecosystem was written fully independently of any pre-existing Lasertag software.

I am fairly confident that the hardware can be adapted to other systems as well, as a lot of the core components are modular enough (though this might take some know-how)
But there is no compatibility built-in to my system as of now.