lzrtag/LZRTag #81: OLED i2c support

A7F, 2022-04-17:
Hi, first thing first, awesome project!
I was wondering if this project does support i2c diplays. Asking because they are super cheap and they should also be easy to add into guns and eventually, rifles! They might display a bunch of information more efficiently rather than relying on haptic and light feedback.

Thank you once more for sharing! :)

Xasin, 2022-04-17:
Hi A7F! 
Thanks for the question.

The ESP32 and my code library do have support for I2C OLED Panels, and I did, in fact, leave out a terminal connection that should be compatible with those cheap 128x32 OLED strip breakout boards you can find on eBay
The connector exposes power and I2C, so it works for all I2C OLED displays.

The code itself does not yet have a OLED panel integration, and I am rather busy with a new internship, so I am currently not actively maintaining the project.
However, adding support should be fairly easy - [you can find my OLED lib code here](https://github.com/XasWorks/XasCode/tree/master/ESP32/LittleOLED)
It's not the best OLED code and you can feel free to use something else, but it works well enough for some basic text and bitmap drawings.

In fact, I am aiming to add proper support for OLED Panels in the next big revision of the Lasertag project, perhaps even using transparent OLED panels as heads-up-display.
You could look through them to aim, they'd display a reticle as well as vital information! 

Again, right now I am not in the right place to actively maintain it, and the component shortage is making it too volatile for me to commit to a new PCB revision since parts can be out of stock for extremely long times.
But I will eventually want to get back to this.

A7F, 2022-04-17:
amazing, thank you. I might give this project a shot (...got it? hehe), and eventually I might add this feature as I am planning to do everything with rifles instead of guns - they look more badass and give a "halo-like" feel.

Just one more question though: is it possible to also add an addressable 5V led strip instead of using embedded rgb leds?

Xasin, 2022-04-17:
Good luck! 
I really hope that the parts for the project are in stock, otherwise it'll require a bit of adjusting. The main concern here is the battery charger, potentially stuff like the accellerometer and speaker IC, etc.

Also yes it absolutely is!
The internal LEDs are all WS2812 LEDs, and the board exposes the data line on the end connector towards the vest.
It should be very easy to chain extra LEDs on to this, and I think I even programmed the LED code so that it can handle arbitrary length LED strips... I am not sure about that though.

But TL;DR, yes. WS2812 LEDs are supported.