lzrtag/LZRTag #82: Cloning the master branch recursively fails due to a missing submodule

InterstellarDrift, 2022-07-22:
Attatched is the relevant section of git output
```
git clone https://github.com/lzrtag/LZRTag --recursive

.....

Cloning into '$HOME/Programs/LZRTag/ESP32/Main/sounds/lzrtag-sfx'...
ERROR: Repository not found.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:lzrtag/lzrtag-sfx.git' into submodule path '$HOME/Programs/LZRTag/ESP32/Main/sounds/lzrtag-sfx' failed
Failed to clone 'ESP32/Main/sounds/lzrtag-sfx'. Retry scheduled
Cloning into '$HOME/Programs/LZRTag/XasCode'...
remote: Enumerating objects: 5231, done.        
remote: Counting objects: 100% (979/979), done.        
remote: Compressing objects: 100% (574/574), done.        
remote: Total 5231 (delta 501), reused 781 (delta 329), pack-reused 4252        
Receiving objects: 100% (5231/5231), 3.42 MiB | 1.57 MiB/s, done.
Resolving deltas: 100% (2785/2785), done.
Cloning into '$HOME/Programs/LZRTag/ESP32/Main/sounds/lzrtag-sfx'...
ERROR: Repository not found.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:lzrtag/lzrtag-sfx.git' into submodule path '$HOME/Programs/LZRTag/ESP32/Main/sounds/lzrtag-sfx' failed
Failed to clone 'ESP32/Main/sounds/lzrtag-sfx' a second time, `aborting`
```
Note: I edited the paths to hide my home directory.

I also searched Github for lzrtag-sfx and found nothing, is it possible this repository is private or was deleted accidentally?

Xasin, 2022-08-24:
Hi! 

Sorry for the delayed response, I was busy with work ^^'

You're right, the directory lzrtag-sfx is private. It was meant to be only temporarily so, as I am using sound effects from a Unity asset package and am not sure if I can have it public.
However, since I am not providing the raw audio game files, I am merely providing "translated", lower quality, encoded files, purely to be used for the Lasertag project, I believe I can open the repository.

Give it another go at pulling!

InterstellarDrift, 2022-08-24:
Thanks! I recloned the repo and it succeeded. Now I'll see if this makes my set of boards work.

closed 2022-08-24 by InterstellarDrift